mvn23/pyotgw #1: Support for tcp/ip connection

pawelka, 2018-10-26:
Is it possible to support for tcp/ip connection?

Some gateways are design to work with nodemcu, so easy remote access is supported instead of serial. 

I'm interested to use gateway with home assistant. I wanted to test it, but tcp/ip support is missing for me.

Mediacj, 2018-10-27:
Yes that would be great but first thanks for your work! 
I am using Esp-link via tcp/ip and I can with succes set the setpoint from within HA but besides that no values come to HA, I get this error by the start of HA:

Home Assistant: 0.81.0

Log Details (ERROR)
Sat Oct 27 2018 21:26:09 GMT+0200 (Midden-Europese zomertijd)

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.6/site-packages/homeassistant/components/opentherm_gw.py", line 75, in connect_and_subscribe
    await gateway.connect(hass.loop, device_path)
  File "/srv/homeassistant/lib/python3.6/site-packages/pyotgw/pyotgw.py", line 52, in connect
    await self.get_reports()
  File "/srv/homeassistant/lib/python3.6/site-packages/pyotgw/pyotgw.py", line 165, in get_reports
    ret = await self._wait_for_cmd(cmd, value)
  File "/srv/homeassistant/lib/python3.6/site-packages/pyotgw/pyotgw.py", line 690, in _wait_for_cmd
    loop=self.loop)
  File "/usr/lib/python3.6/asyncio/tasks.py", line 362, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError

koen01, 2018-10-29:
I'm seeing the exact same error in my log file. using this:

`opentherm_gw:
  device: socket://192.168.1.101:6638
  climate:
    name: Thermostaat
    precision: 0.5
  monitored_variables:
    - room_setpoint
    - room_temp`

mvn23, 2018-10-30:
TCP/IP support is indeed already built in, but there are reports of connection issues in combination with NodeMCU. Please see home-assistant/home-assistant#17263 for progress on that. I've ordered an ESP8266 to run some tests.
I am closing this issue as TCP/IP is already supported.

closed 2018-10-30 by mvn23