mvn23/pyotgw #2: "[pyotgw.protocol] Disconnected: None" crashes homeassistant

wavemaking, 2019-01-11:
pyotgw (as part of the opentherm_gw component) works partially for me. It works from the dev version of homeassistant (not from docker or the python install). I get updates from the monitored values and I can control the target temperature. But inevitable after some time homeassistant freezes with the following last messages in the log:

`2019-01-10 12:20:19 DEBUG (MainThread) [pyotgw.protocol] Watchdog triggered!
2019-01-10 12:20:19 DEBUG (MainThread) [pyotgw.protocol] Canceling Watchdog task.
2019-01-10 12:20:19 DEBUG (MainThread) [pyotgw.pyotgw] Scheduling reconnect...
2019-01-10 12:20:19 DEBUG (MainThread) [pyotgw.pyotgw] Reconnecting to serial device on socket://192.168.71.103:6638
2019-01-10 12:20:19 ERROR (MainThread) [pyotgw.protocol] Disconnected: None
`

I note that I also have many repetition of these messages:
`2019-01-11 01:33:26 WARNING (MainThread) [pyotgw.protocol] Command PR failed with SE, retrying...
2019-01-11 01:33:26 DEBUG (MainThread) [pyotgw.protocol] Got possible response for command PR: SE`
The component works regardles of these messages, so I am not sure if it is relevant.

maikelwever, 2019-03-11:
Same problem for me: after this happens, HomeAssistant is not reachable via the WebUI anymore, and one core of the system it's on is pegged to 100% usage.

Installed versions of libraries:
pyotgw==0.4b1
homeassistant==0.89.1
pyserial==3.4
pyserial-asyncio==0.4

The logs I got look the same:
```
Mar 11 08:34:24 hass hass[10783]: 2019-03-11 08:34:24 DEBUG (MainThread) [pyotgw.protocol] Watchdog triggered!
Mar 11 08:34:24 hass hass[10783]: 2019-03-11 08:34:24 DEBUG (MainThread) [pyotgw.protocol] Canceling Watchdog task.
Mar 11 08:34:24 hass hass[10783]: 2019-03-11 08:34:24 DEBUG (MainThread) [pyotgw.pyotgw] Scheduling reconnect...
Mar 11 08:34:24 hass hass[10783]: 2019-03-11 08:34:24 DEBUG (MainThread) [pyotgw.pyotgw] Reconnecting to serial device on socket://opentherm-gw-esp.skynet.lan:6638
Mar 11 08:34:24 hass hass[10783]: 2019-03-11 08:34:24 ERROR (MainThread) [pyotgw.protocol] Disconnected: None
```

maikelwever, 2019-03-11:
How to reproduce this issue:

- Opentherm gateway with esp8266 as serial<->wifi bridge, in my case using easy-esp.
- Install home assistant
- Configure the opentherm_gw component to connect to the esp8266

When connected, do either of these:

- Reset the esp8266
- Pull the power from the opentherm gateway and/or the esp8266
- Open a new connection to the serial bridge, this causes previous connection to be dropped

How to observe the bug afterwards:

- Open htop, task manager, or whatever is your favorite resource monitor, and notice one Python process (home assistant) using 100% cpu core but not actually doing anything useful.
- Home assistant is not reachable.

maikelwever, 2019-03-11:
A little debugging shows the code gets stuck here:

https://github.com/mvn23/pyotgw/blob/master/pyotgw/protocol.py#L64

andriej, 2019-03-12:
Can't wait for the fix to be pushed to HA - this issue killed my HA today. :-)

mvn23, 2019-03-12:
Thanks for the report and PR, I've been rather busy the last few months but I will have a look at it this week.

mvn23, 2019-03-12:
Fixed in #5

closed 2019-03-12 by mvn23

andriej, 2019-03-14:
I'm after first dropout of connection with updated (via pip install manually) pyotgw lib.
It seems to work, HA is not consuming whole CPU and data seems to be updated (so the reconnection was successfull so far):
[image]