mvn23/pyotgw #3: Causing unstable status of HA?

andriej, 2019-01-21:
Since I use native HA integration - after latest fix - it works (I mean it reconnects and I'm able to control my thermostat) but after running for many hours HA seems to get more and more unstable and eventually loosing some socket integrations like Xiaomi Aqara.

Is there possibility that there's some memory leak or socket's limit is getting eaten by component?
I'd like to help, just let me know how can I participate.

gdschut, 2019-01-21:
My Influxdb was causing a lot of memory peaks. Perhaps you can try to stop it for a while to see what happens?

andriej, 2019-01-22:
Now darksky have issue to connect, with error:
`(Caused by NewConnectionError(': Failed to establish a new connection: [Errno -2] Name or service not known',))`

May it be cause by other async component?

gdschut, 2019-01-22:
I'm using darksky also, but without any errors right now. What is an async component?

gdschut, 2019-01-23:
Thats interesting! I never had a config for OTGW with MQTT, but use MQTT already for other sensors. How to change to MQTT?
For now I have inserted an automation that reloads all automations at night, hope this works for now...

gdschut, 2019-01-30:
Made a move to OTGW via MQTT using TCP branch of [https://github.com/jodur/py-otgw-mqtt/tree/tcp](https://github.com/jodur/py-otgw-mqtt/tree/tcp). This is a python implementation, so perhaps it is usefull for migrating tot HA?
Also moved almost all of my automations to Node-Red, resulting in a very stable environment again. If there is any development on the OTGW component for HA, I am still interested, but for now I prefer a stable central heating system, without getting my wife upset about home automations... Thanks anyway for your effort.

andriej, 2019-03-08:
Just getting back to component - I'm on 0.89.1 and came back here to check how things are going. :-)

mvn23, 2019-03-12:
This could be a result of connections not being cleaned up on disconnect. May be fixed by #5 so please check if the problem persists with the next version (will be released soon).

andriej, 2019-03-20:
I still had problem even after fixes - less with HA, more with wifi I think - a lot of errors and reconnections. 
Today I've just switched both to 0.90 (which got 3 fixes found meanwhile) and to Ethernet adapter to OTGW (so no more goddamn disconnections). 

Works smooth so far, will monitor.

andriej, 2019-03-26:
Just an update - for me it seems to be already fix after moving from ESP-nodemcu to LAN-gateway and fixes are waiting for next HA release. 
Probably issue to be closed.

mvn23, 2019-03-28:
Is this still present in 0.4b2 (HA 0.90.2)?

gdschut, 2019-03-28:
I m currently on MQTT communication, but hope to switch back to native OTGW as soon as I have time for it to do and also time for testing and eventualy rollback if it is not.... Right now I have other connection problems as well from other components so perhaps it would be better to wait for a stable situation...

westenberg, 2019-03-28:
Until now I have no more OTGW issues anymore with HA 0.90.2

andriej, 2019-03-28:
@gdschut those problems could be caused by unstable integrations. I had the same and worked out together with author of component few fixes that made it stable :-)
Was also using MQTT earlier.

andriej, 2019-04-02:
I'm closing this issue as all of the issues disappeared with both hard work with debug and code from mvn23 and change on connectivity side with fixed LAN adapter. 
@gdschut make sure you try the latest versions of HA while switching back to OTGW component.

closed 2019-04-02 by andriej