mvn23/pyotgw #4: Component doesn't recognize termostat value?

andriej, 2019-03-10:
I've came back to the native integration in HA to see if anything changed in latest release.
It was running stable so far, but I have iSense thermostat in which there's a schedule.

[image]

And while schedule changes the value, I can't see it in HA - I can only see that boiler is heathing and when I click on it I can see the value.
Also what I can see is that the 'set' value is changed for a split second for the value from thermostat and come back to 22.5 (set previously as a setback in HA and in termostat). Should show 24.5, which is shown on the graph.

mvn23, 2019-03-12:
I don't have a thermostat with scheduling function, so I can't really debug this myself. I will implement some extended debugging in the next release to trace this issue.

maikelwever, 2019-03-12:
I also have a Remeha iSense (v19), with scheduling.

I think @mvn23 is correct in thinking this issue lies with the HA component, since I can see the correct values in the history graphs. I'll take a look at the component in HA and see if I can improve/fix this.

andriej, 2019-03-13:
Yes, I forgot to point my assumption that it may be frontend issue also.
I've just bumped version to latest (to fix the hanging issue).

What I wanted to avoid is situation also like this one:
[image]

mvn23, 2019-03-18:
Those vertical lines are caused by the climate platform initializing its temperatures to 0, which is fixed in home-assistant/home-assistant#22112
Closing this as the original issue is not a bug in pyotgw.

closed 2019-03-18 by mvn23