mvn23/pyotgw #8: Mismatched data readings?

andriej, 2019-08-29:
I'm using OTGW integration in HA and it just works.
Lately I've been updating the UI and I've encountered something that I assume is not right.
Names are (yet) default from integration, might be easier to debug:

[image]
[image]
[image]

Currently I've disabled the central heating as it's still summer and it's not needed. 
It can be seen that "Central Heating" (picture 1) pump starts counter is increasing while "hot water pump starts" (picture 3) is flat.
Shouldn't it be switched? We use hot water only so far. :-)

I'm wondering if it's fault of boiler or mismatch somewhere in the code.

mvn23, 2019-08-31:
I see the same behavior here. However, after checking the code, I found that it matches the protocol specification I used (v2.2): message ID 117 (0x75) for CH pump starts and 118 (0x76) for hot water.

It is unlikely that this was changed between versions, so either the specification I have is incorrect or both of our boilers have implemented this the wrong way around. If I can find some proof for the former I will update the code accordingly.

mvn23, 2021-03-11:
No progress in more than a year. While this bug is backed up by real-world examples in at least 2 cases (including my own), we do follow the OpenTherm spec with the current implementation.
Other software packages seem to deal with this in the same way. While this in itself is of course no valid reason to not change anything, I do want to mention it as an additional consideration.
Closing this for now, please feel free to open a new issue if and when new information becomes available.

closed 2021-03-11 by mvn23