mvn23/pyotgw #9: [HA] OpenTherm integration

andriej, 2019-09-20:
Would it be possible to implement OpenTherm Gateway integration in Home Assistant as an "config flow" one?
I'm already happy with it - but additional possibilities hidden behind entity register, disabling those which are not used (supported by boiler) would be even better.

Just a thing to consider - as I can see OT is already very integrated inside HA anyway, hope it's not much of a hassle to make integration via UI possible.

andriej, 2019-09-21:
Looking more into the issue - to be able to turn off entities in HA they only need to have unique id set by integration. So even (less) hassle getting it to work and so much easier life with possibility of turning unwanted ones off forever.

mvn23, 2019-09-21:
Closing this as it is not a pyotgw issue.
Either way, I am planning to do this in the future.

closed 2019-09-21 by mvn23