mvn23/pyotgw #11: Timed out waiting for command: TT, value: 14.0.

basnijholt, 2020-01-31:
I keep getting the following error in HA `0.104`:
```
Timed out waiting for command: TT, value: 14.0.
```

This is when setting the temperature to 14, which doesn't happen.

This is proceeded by the following (large, sorry) excerpt of my log
```
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Added line 4098 to command queue. Queue size: 1
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Got possible response for command PR: PR: O=T17.00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 19 39 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 63 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: R 00 76 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 76 99 b8
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 19 39 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 31 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 31 4b 14
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 19 38 e6
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 05 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 05 00 ff
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 19 38 e6
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 1b 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: R 00 77 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 77 d9 a8
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 19 38 e6
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 0e 64 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 0e 64 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 00 02 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 01 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 05 01 06 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: B 04 11 00 00
2020-01-31 10:15:22 DEBUG (MainThread) [pyotgw.protocol] Processing: T 00 19 00 00
[... 2046 lines cut ...]
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Received line 4720: T10010600
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Added line 4720 to message queue. Queue size: 2878
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Received line 4721: BD0010600
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Added line 4721 to message queue. Queue size: 2879
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:09 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Received line 4722: T00110000
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Added line 4722 to message queue. Queue size: 2880
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Received line 4723: BC0110000
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Added line 4723 to message queue. Queue size: 2881
2020-01-31 10:20:10 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Received line 4724: T80190000
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Added line 4724 to message queue. Queue size: 2882
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Received line 4725: BC0193199
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Added line 4725 to message queue. Queue size: 2883
2020-01-31 10:20:11 DEBUG (MainThread) [pyotgw.protocol] Watchdog reset!
2020-01-31 10:20:12 ERROR (MainThread) [pyotgw.pyotgw] Timed out waiting for command: TT, value: 14.0.
```

In the log, the queue size goes from 1 suddenly to 2444...

Not sure how to debug this. Could you perhaps give me some suggestions?

mvn23, 2020-01-31:
Thanks for the report.
Before diving into the log analysis, can you please report the following:
* How are you connecting to the gateway? USB/TTL or wired/wireless network? In case of a network connection, please specify the full path (e.g. wired from HA to router, then wireless to gateway)
* What hardware and os are you using to run Home Assistant?

From the logs, it looks like there's an issue with the message processing task.
There are 2 queues handled by separate tasks: the command queue for everything command-related and the message queue for all protocol messages. The command queue is the lesser used one and seems to be handled fine (size remains 1). The message queue however is already `2443` items long at the start of the logs, then increases to `2477` at `10:15:38` where a batch of messages is processed, dropping the size to `2310`. After that it only increases to `2883` in the end.
At approximately 1 message per second, this means that the first (oldest) message in the queue is now more than 48 minutes old.

The fact that the queue size decreased in size halfway through indicates that the queue processing task is still running. It seems to be blocked most of the time though. This does not necessarily mean that the problem lies with pyotgw or the OpenTherm Gateway integration in Home Assistant. If you're willing to tinker a bit to find the issue, the first thing you could try is to disable all other integrations to see if the problem persists. If it does, post another debug log of **all** components, not just `pyotgw.*`.
If the problem disappears, enable the other integrations one by one to see when the issue occurs again.

To make your own life a bit easier, you can make a backup of your config dir (and database if separate) first to restore later.

basnijholt, 2020-01-31:
Thanks for the fast response!

* I run HassOS with Hass.io on a Raspberry Pi 4
* I am using a NodeMCU with an OTWG from [Nodo-Shop](https://www.nodo-shop.nl/nl/opentherm-gateway/188-opentherm-gateway.html), the NodeMCU is connected over WiFi to my router and my RPi is connected via an ethernet cable to the router.

I am definitely willing to help to debug this and tinker.

I will try to run everything on a different HA installation where I do not have any integrations tonight!

basnijholt, 2020-02-01:
@mvn23, I have moved the integration to a Hass.io, RPi 3, without any other integrations running.

I am still experiencing many problems. 

For example from the HA UI logs:
```
Timed out waiting for command: PR, value: C.
10:49 AM __main__.py (ERROR) - message first occurred at 10:45 AM and shows up 3 times
Disconnected: connection failed (socket disconnected)
10:49 AM __main__.py (ERROR) - message first occurred at 10:43 AM and shows up 25 times
Timed out waiting for command: PR, value: B.
10:49 AM __main__.py (ERROR) - message first occurred at 10:43 AM and shows up 6 times
Timed out waiting for command: PR, value: A.
10:49 AM components/opentherm_gw/__init__.py (ERROR) - message first occurred at 10:43 AM and shows up 3 times
Command PR failed with PR: A=OpenTherm Gateway 4.2.5, retrying...
10:49 AM __main__.py (WARNING)
Unknown message in command queue: PR: A=OpenTherm Gateway 4.2.5
10:49 AM __main__.py (WARNING)
Timed out waiting for command: PR, value: T.
10:49 AM __main__.py (ERROR) - message first occurred at 10:44 AM and shows up 6 times
Timed out waiting for command: PR, value: S.
10:49 AM __main__.py (ERROR) - message first occurred at 10:44 AM and shows up 14 times
Timed out waiting for command: PR, value: R.
10:49 AM components/opentherm_gw/__init__.py (ERROR) - message first occurred at 10:43 AM and shows up 10 times
Timed out waiting for command: PR, value: P.
10:48 AM components/opentherm_gw/__init__.py (ERROR) - message first occurred at 10:43 AM and shows up 6 times
Timed out waiting for command: PR, value: O.
10:46 AM __main__.py (ERROR)
Timed out waiting for command: PR, value: G.
10:45 AM __main__.py (ERROR)
Timed out waiting for command: PS, value: 0.
10:44 AM __main__.py (ERROR)
Timed out waiting for command: PS, value: 1.
10:44 AM __main__.py (ERROR)
```

Then the full log (36 MB) can be found [here](http://files.nijho.lt/home-assistant.log).

I don't know how to proceed. 

I have also found [this](https://community.home-assistant.io/t/opentherm-otgw-node-mcu-unreliable/158813) post on the community forum, which might be a related problem.

mvn23, 2020-02-01:
The main issue now is that there are a lot of connection drops/interruptions `ERROR (MainThread) [pyotgw.protocol] Disconnected: connection failed (socket disconnected)`. All other error messages are related to that. This problem differs from the one in the first post.
I've seen similar reports a few times before when using wifi to connect to the gateway, but I was never able to reproduce it myself with a similar setup. In your case however, the extremely regular interval between disconnects (16 seconds every time) stands out as unusual to me.
Unfortunately I am not aware of a way to get better control over the underlying connection (pyserial is handling that).

basnijholt, 2020-02-06:
The only thing I had running on that RPi3 was https://www.home-assistant.io/integrations/dsmr/ which uses `dsmr_parser` that also uses [`https://github.com/ndokter/dsmr_parser/blob/master/setup.py#L13`](https://github.com/ndokter/dsmr_parser/blob/master/setup.py#L13), maybe that somehow interfered, however, I think I had it disabled...

I have another Pi on which I can try to install HA without anything else. I will do that tonight to see if I can learn anything more.

I am relatively sure it's a software issue. After I restart HA, the `opentherm_gw` integration starts to work again. However, after some time, I keep running into `Timed out waiting for command: TT, value: SOME_TEMP_HERE` again (on my original RPi4, main instance).

basnijholt, 2020-02-06:
I've uploaded my log [here](http://files.nijho.lt/home-assistant-queue.log) BTW.

mvn23, 2020-02-07:
The command timeout defaults to 3 seconds. This should be enough, but for testing purposes you could try to increase it a bit. The easiest way to change it globally would be to increase the `OTGW_DEFAULT_TIMEOUT` in `vars.py`.
However, from the log it looks like either some commands don't get a reply at all, not even after the timeout has expired. So either the command does not get sent at all or one of the messages (either command or response) gets lost somewhere. I may consider applying the retry logic to timeouts as well, but that feels more like a workaround than an actual solution...
When I get back home I will try to reproduce once more, this time with HA running on a pi rather than my normal server.

basnijholt, 2020-02-07:
I do have problems, no matter where I run the integration.

I have set up a completely new Home Assistant install (former Hass.io) on a RPi3.

I am now trying to install the `opentherm_gw` as `custom_integration` to change the timeout.

basnijholt, 2020-02-08:
Using a completely fresh HA install with only `default_config:` and `logger` in my configuration, I am able to reproduce the original error messages (by just waiting and changing the temperature of few times).

I have doubled `pyotgw`'s timeout, however, it doesn't seem to help at all.

[Here are the log-files](http://files.nijho.lt/home-assistant-longer-timeout.log), I wish I would know where to start.

I did get
```
Unknown message in command queue: Low power
``` 
and
```
Unknown message in command queue: Medium power
```
once.

andriej, 2020-02-08:
Have you check the ping/signal strenght of the gateway?
I had a lot of problems with messages and they are all gone since I've moved to wired connection.
OTGW issues lots of messages every second, while WiFi may have it's hiccups.

basnijholt, 2020-02-08:
I do not think it's related to the signal's strength.

I am currently trying `otmonitor` and experience no problems with that.

@mvn23, I don't think I am the only one experiencing these problems, [this](https://gathering.tweakers.net/forum/list_message/60980688#60980688) (in Dutch) person has a similar setup to mine and has the same problems. He fixed it by going to the following route "OTGW -> OTMonitor -> OTMonitor MQTT -> MQTT Broker -> HA" instead of the default integration.

mvn23, 2020-02-09:
> I have doubled `pyotgw`'s timeout, however, it doesn't seem to help at all.

I expected as much. Messages seem to actually get lost somewhere. We will have to analyze network traffic to find out where exactly.

> I did get
> 
> ```
> Unknown message in command queue: Low power
> ```
> 
> and
> 
> ```
> Unknown message in command queue: Medium power
> ```
> 
> once.

I've seen this in logs before, but my own setup does not have these messages (maybe not supported by my boiler). It's unrelated to the timeouts and I expect it's a way to communicate the fact that the boiler's power mode has changed. Could you test and verify this? Then we can implement these messages as well.

> I do not think it's related to the signal's strength.
>
> I am currently trying `otmonitor` and experience no problems with that.
>
> @mvn23, I don't think I am the only one experiencing these problems, [...]

There have indeed been similar reports in the past. So far I've never been able to reproduce, but it always seems to be related to WiFi versions of the gateway (or WiFi somewhere else in the link). However, even with WiFi, any failed transmissions/missing packets should normally be corrected on the TCP level.
As for `otmonitor`, it's definitely a good option. But if the issue is in the network connection then `otmonitor` will have the same problems, though it may be better at hiding/correcting them.

I will hook up a NodeMCU to my gateway this afternoon and run a Home Assistant instance on a raspberry pi to see if I can reproduce. Are there any specific steps I could take to trigger the issue?

basnijholt, 2020-02-09:
The otmonitor tool reports that it has 0 errors, so I have the impression that it doesn't encounter them. Unless it silently corrects errors. 

To get the errors I don't do anything else than changing the set temperature a few times and waiting.

mvn23, 2020-02-10:
What firmware do you use on the NodeMCU? I am using `ESP_Easy_mega-20200204_normal_ESP8266_4M1M` at the moment without any issues, even when changing the target temp a few times. Uptime is only 1.5 hours so far, but I will leave it running overnight to see if anything happens.

The setup I use for testing is a raspberry pi 2B (raspbian, venv, default config with only `opentherm_gw` added), connected via ethernet to my IOT network. From there to the gateway I have a wireless connection to a NodeMCU, which is connected with some dupont wires to the TX/RX of the gateway.

mvn23, 2020-02-11:
It has been running like above for 20 hours now. In the mean time I have changed the temperature several times, both from HA and from the thermostat.
In these 20 hours I have logged 5 disconnects, all of which were successfully recovered. I have seen no timeouts at all and the maximum amount of backlogged messages in the queue was 11.
This is on a pi 2B, which is not even recommended for production HA installs anymore. A pi 3 or 4 should only perform better with regards to message processing.
Apart from the firmware on the NodeMCU, is there anything else I should change to be able to reproduce?

basnijholt, 2020-02-16:
I have made HassIO add-on with [`otmonitor`](https://github.com/basnijholt/addon-otmonitor) and it has been running great over MQTT for a week now.

However, I do prefer the native implementation.

Since I have no connection errors at all using `otmonitor` it leads me to believe that it's an actual bug with `pyotgw`. I will give the other firmware a try though, as I am running ESPEasy R147 on my NodeMCU.

basnijholt, 2020-02-16:
Unfortunately, I am not able to upgrade my R147 firmware to the `mega` pre-release OTA :(

I (believe) I would have to take the NodeMCU out of its case and do it via a cable which I do not have the time for ATM.

mvn23, 2020-02-16:
The earliest opportunity for me to try R147 is next weekend. In the mean time I will look into more debugging output when a connection reset is triggered.
I am aware of the fact that otmonitor seems to have no issues with the connection but so far I have neither encountered issues in my own wired LAN setup (although #13 seems to suggest that that happens as well), nor received similar bug reports with direct serial connections. At this point we're going to need a more thorough debugging setup to rule out any network issues and get to the root of the issue.

mvn23, 2020-02-16:
Latest master should give some more insight in the state of the serial connection when a reconnect is triggered.

mvn23, 2020-03-03:
Just tested ESPEasy R147 and I do indeed see erratic behavior on there as well. I suggest you try the `mega` firmware to see if that improves your situation. Please post back here with the results :-)

basnijholt, 2020-03-03:
Oh, that's great news, thanks a lot for trying!

I will try it ASAP and report back 👍

basnijholt, 2020-03-03:
I have just installed `ESP_Easy_mega-20200301_normal_ESP8266_4M1M.bin` an hour ago, an so far I haven't encountered the errors I saw before! :tada:

I wasn't 100% sure about the settings for my NodeMCU, but [this](https://github.com/basnijholt/home-assistant-config/issues/7) is what I have done.

mvn23, 2020-03-03:
> I did get this error twice:
> 
> ```
> Command PR failed with SE, retrying...
> ```
This is probably caused by the serial log messages from ESPEasy. Try setting `Serial Log Level` to `None` in ESPEasy Advanced config.

basnijholt, 2020-03-03:
That indeed seemed to have fixed it! 👍

basnijholt, 2020-03-04:
I did get these error messages since yesterday:
[image]

mvn23, 2020-03-04:
Can you provide a debug log of errors like that? From your screenshot it looks like some data gets lost somewhere.
The main reason I suspect that is the message `00000010/00001100,4PS: 0`. The first part `00000010/00001100,4` is a partial response to a `PS=1` command, which should be followed by more data and finally a `\r\n` line terminator. Instead, `PS: 0\r\n` follows, which is the full response to a `PS=0` command.
What I noticed is that in the ESPEasy Advanced config, your serial port is set to 115200baud. The serial server is set to 9600, which is correct, but I am unsure if these 2 are interfering with each other in any way. Can you try setting the former to 9600 as well (do not disable the serial port, this will affect the serial server) and provide a debug log with that?

basnijholt, 2020-03-05:
I have changed number 115200 to 9600 as well, I had it like that because that's what I read [here](https://www.nodo-shop.nl/nl/index.php?controller=attachment&id_attachment=35).

After that change I still find many errors:
[image]

And I uploaded the logs [here](http://files.nijho.lt/home-assistant.log).

Hope that tells you something.

mvn23, 2020-03-05:
> I have changed number 115200 to 9600 as well, I had it like that because that's what I read [here](https://www.nodo-shop.nl/nl/index.php?controller=attachment&id_attachment=35).

Got it, they probably don't affect each other then.

Thanks for the log. It looks like all commands after `2020-03-04 23:23:39` time out without any response from the gateway. Regular messages keep coming in, so the connections seems OK most of the time.
This could be an issue with the serial transport. Can you confirm with `tcpdump` or something similar that the commands actually get sent? I suspect that they don't get sent at all in your case, which would be a bug in `pyotgw`. If they do get sent but you don't get a response, it's a network/gateway issue.

basnijholt, 2020-03-06:
Unfortunately, the original error started happening again (consistently).
```
Timed out waiting for command: TT, value: 20.0.
```

With logs http://files.nijho.lt/home-assistant-errors-again.log

Could you perhaps tell me how to use `tcpdump` or point me at some documentation?

mvn23, 2020-03-06:
```
tcpdump -w otgw.pcap -i <interface> host <ip> and port <port>
```

Where <interface> is the network interface you're using to communicate with the gateway, <ip> the gateway ip and <port> the gateway port for serial traffic. The dump will be placed in your working directory as `otgw.pcap`. This will capture all traffic to and from the gateway.

mvn23, 2021-02-16:
The last response is now almost 1 year ago and there have been a few new releases since this issue was filed. Please file another bug report if you're still experiencing this issue with the latest release.

closed 2021-02-16 by mvn23

basnijholt, 2021-02-17:
Unfortunately, I never got it to work properly, so instead I am using the official `otmonitor` tool.

For convenience, I have created a Home Assistant Supervisor Add-on: https://github.com/basnijholt/addon-otmonitor

andriej, 2021-02-17:
Kind of offtopic, but I've migrated OTGW integration from direct connection to OTGW->otmonitor addon->gateway and I like it.
I finally have full look-into the messages when I want without need of disconnecting anything (my ethernet-serial clone can only provide data for one client, so I had to disconnect HA when I wanted to see the messages with otmonitor previously).