mvn23/pyotgw #21: Relation to OTMonitor

iamkarlson, 2020-10-08:
Hi, 
I'm trying to figure out how does this whole project works and have some dummy questions. 
1. What is a relation to [OTMonitor](http://otgw.tclcode.com/otmonsrc.html)?
2. What are the minimum requirements for dev env you have to satisfy in order to run the project? 
3. Are there any tests or some testing scripts?

mvn23, 2020-10-08:
1. No specific relation. Both otmonitor and pyotgw can be used to connect to the OpenTherm Gateway. Otmonitor has a GUI and some external connectivity options (web, mqtt) and can be used to update the firmware on your gateway. pyotgw can be used to interface with the gateway from a python environment only.

2. Any python dev setup will work. At present, the aim is to support the same python versions as Home Assistant (3.7.1 or later at the time of writing). It is advisable to install `pre-commit` in your local repository as per the [Readme](https://github.com/mvn23/pyotgw/blob/master/README.md#development).

3. The [Readme](https://github.com/mvn23/pyotgw/blob/master/README.md#usage-example) has a usage example. There is currently no test suite to verify functionality.

iamkarlson, 2020-10-08:
Thanks @mvn23! About the first, so it's totally independent from the otmonitor? I mean I don't need any of its dependencies to run it, right?

mvn23, 2020-10-08:
Indeed, you only need python and a way to connect to the OpenTherm Gateway.
pyotgw uses [pyserial-asyncio](https://github.com/pyserial/pyserial-asyncio), which in turn relies on [pyserial](https://github.com/pyserial/pyserial) to make the connection, so you can check their documentation for [supported connection types](https://pythonhosted.org/pyserial/url_handlers.html).

closed 2020-10-09 by iamkarlson