mvn23/pyotgw #26: "[pyotgw.protocol] Disconnected: None"

DJTerentjev, 2021-01-12:
Thank you for developing this important for me  library and integration. Could you please have a look at problem I face. I have seen issue like this closed in the past, but I have this error with newest version of Home Assistant . I think this issue caused by network, but it strange for me that a little delay in network lead to boiler to change state (to off).  Even restart of Home Assistant  when it defiantly disconnected for several minutes  keep same boiler state. I have it not very often each 3 to 10 days, but it make me upset when I found boiler off. Please see log attached: [https://pastebin.com/v2SNhgTM](url)

My be it is possible, just by pass this error without changing boiler state.

mvn23, 2021-01-13:
There is only debug logging from the Home Assistant component in the log. No debug info from the `pyotgw` library, only a single error message.
All I can tell from these messages is that there is definitely a disconnect happening. `Disconnected: None` indicates that a regular EOF is received, or the connection was aborted or closed by the local side of the connection. This may be caused by the built-in watchdog detecting inactivity on the connection and triggering a reconnect attempt. Enable debug logging for `pyotgw` to get more information on this.
If it is indeed the watchdog then there is probably an issue with the network connection. Make sure there is nothing else connecting to the gateway.
There have also been reports of similar issues when connecting to the OpenTherm Gateway via wifi, especially when using ESP-based wifi controllers.

DJTerentjev, 2021-01-13:
Right I am using ESP-based wifi controller. And I am using it for two years. Actually I was planning to use wired connection first, but I was not able to setup TTL to Ethernet adapter that time and decided to try with ESP. And it was working fine. Only after I update to version with on/off boiler feature (thank you for this feature ) I start noticed my boiler go off time to time.   It  is reconecting back so everything looks normal, just boiler is off. Please let me try to get pyotgw log and may be you can help to keep boiler state.

DJTerentjev, 2021-01-13:
Please see my log: [https://pastebin.com/PkrWP8Lw].
I see Watchdog  was triggered and then process of reconnect. After reconnect there is line:
`Sending command: CH with value 0`
Should be send previous state (which was 1)?

mvn23, 2021-01-13:
There is nothing within `pyotgw` actively changing the `CH` value upon disconnect/reconnect. `pyotgw` is only a layer between python and the OpenTherm Gateway. If the `CH` command is sent from your Home Assistant instance, it is because the [`opentherm_gw.set_central_heating_ovrd`](https://www.home-assistant.io/integrations/opentherm_gw/#service-set_central_heating_ovrd) service is called. However, a gateway disconnect event does clear all other `opentherm_gw` sensors and binary sensors, which may cause your automations to trigger this behavior.

For the disconnects, this has been reported in the past with ESP-based wifi connections. Unfortunately I have never been able to reproduce the issue, despite extensive testing with various ESP8266 dev boards. Then there is the fact that reconnects are handled as expected, leading me to conclude that this is a network issue.

In short, sending the `CH` command is never triggered by `pyotgw` itself and there is no indication that the disconnects are caused by a bug in `pyotgw`. Nothing here is pointing to an issue with `pyotgw`, so I'm closing this issue.
For further help with your automations and scripts I would like to kindly refer you to the [Home Assistant Help page](https://www.home-assistant.io/help/). The Forums and Discord community can be a great resource when running into trouble.

closed 2021-01-13 by mvn23