mvn23/pyotgw #29: deprecation warning spam in log

andriej, 2021-02-09:
While browsing local log files on HA with OTGW integration, I've encountered this deprecation warning spam in log:
```
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
defa3e4e[597]: /usr/local/lib/python3.8/site-packages/pyotgw/protocol.py:147: DeprecationWarning: The loop argument is deprecated since Python 3.8, and scheduled for removal in Python 3.10.
defa3e4e[597]:   await asyncio.sleep(timeout, loop=self.loop)
```

mvn23, 2021-02-09:
Thanks for the report, this will be fixed in the next version after firmware 5.0 support and some other things are done.
The only thing I don't understand is why I'm not seeing this anywhere on my setup :confused:

andriej, 2021-02-09:
On Debian 10 it's in /var/log/messages by default.
Thanks for the fix, will implement later on :)

mvn23, 2021-02-09:
Did some digging, turns out they are usually only shown when python is running in [dev mode](https://docs.python.org/3/library/devmode.html#devmode), which is not the case on my production environment.

andriej, 2021-02-09:
Neither is mine - it's standard Supervised installation :-)

tirkarthi, 2021-03-07:
The loop parameter was removed in Python 3.10 : https://bugs.python.org/issue42392

commit 4b429b16 (2021-03-10): Remove loop arguments that were deprecated in python 3.8, removed in python 3.10 (fixes #29)
closed 2021-03-10 by mvn23