mvn23/pyotgw #34: Command PR with value D raised exception: Bad Value:

DJTerentjev, 2021-09-22:
Hello, not an issue, just a question. 
I am using OTGW integration for long time. It's work great! Only every start-up of HA I got error: 
Command PR with value D raised exception: Bad Value: The command contained a data value that is not allowed or not supported on this version of the OpenTherm Gateway.

As I understand it some command from boiler:  Print Report: The DS18x20 temperature sensor function. Which not suported by OTWG.

Is something I can do to get rid of this error?

Thank you!

mvn23, 2021-09-22:
Thank you for the report. This may indeed be a bug in pyotgw. `PR=D` was added in Opentherm Gateway 5.0 and the error should be silently ignored if your version is lower than that.

Can you confirm your Opentherm Gateway firmware version?

DJTerentjev, 2021-09-22:
I think it was 4.2.4.

mvn23, 2021-09-22:
If you enable entity `sensor.otgw_about_gateway_*` (* = your gateway ID) it will show you the version.

DJTerentjev, 2021-09-22:
OpenTherm Gateway 4.2.5

mvn23, 2021-09-22:
Thanks, will be fixed in v2.0.

DJTerentjev, 2021-09-22:
Thank you! 
One more question if you don't mind.
Please see below. All entities related to boiler errors have unavailable state.  I older version of integration I had them showing states. And once I catched error of the boiler one. It is because errors inactive at the moment? 
[image]

mvn23, 2021-09-23:
If the sensor is marked as "unavailable" it means that the Opentherm Gateway did not (yet) supply any useful information on that particular parameter, i.e. the state is unknown.

DJTerentjev, 2021-09-23:
Got it. Thank you!

andriej, 2021-09-23:
I was about to post same issue but thought it's my installation fault. 

Is it worth upgrading to 5.0? :-)

commit 429d5a6b (2022-07-15): Check firmware version before sending `PR=D` (fixes #34)
commit c293abdb (2022-07-15): Check firmware version before sending `PR=D` (fixes #34)
closed 2022-07-15 by mvn23