mvn23/pyotgw #46: Room Setpoint is 0

PierreAronnax, 2022-11-12:
After updating Home Assistant to 2022.11.2 the Room Setpoint is mostly 0
Likely due to #44

Log from Opentherm Monitor:
```
11:51:10.945745	T10101200	Write-Data	Room setpoint (MsgID=16): 18.00
11:51:10.997147	B70100000	Unk-DataId	Room setpoint (MsgID=16): 0.00
11:51:11.016209	AD0100000	Write-Ack 	Room setpoint (MsgID=16): 0.00

11:53:15.372293	T10101200	Write-Data	Room setpoint (MsgID=16): 18.00
11:53:15.422900	B70100000	Unk-DataId	Room setpoint (MsgID=16): 0.00
11:53:15.467431	AD0100000	Write-Ack 	Room setpoint (MsgID=16): 0.00
```

mvn23, 2022-11-12:
Thank you for the report. It is indeed caused by #44, combined with the fact that the gateway apparently forwards `Unknown Data ID` messages from the boiler as `Write Ack` messages. I have made an [inquiry](https://www.domoticaforum.eu/viewtopic.php?t=13778) (awaiting mod approval) about this behavior, and will fix this problem after I get feedback from there to make sure the fix makes sense.

PierreAronnax, 2022-11-14:
This behavior is specific for Room Setpoint. From gateway.asm:
```
We don't want an überclever thermostat to stop sending these
messages, so make sure we always return an acknowledgement
```
I agree that it should acknowledge the received value from the thermostat, however you cannot require that users upgrade to  new (not yet exising) gateway firmware after updating a library, for functionality that used to work before.

I think it is safe to always ignore 0 from both directions, since 0 also indicates no remote override is to be applied.

Also, shouldn't any fix be duplicated to `MSG_TRSET2`?

mvn23, 2022-11-14:
Thanks for the additional info. I wanted to be sure I understood the behavior before preparing a fix, and with this information I do.
Either way the fix would have been backwards compatible, but with the info from gateway.asm we can safely assume it will not change in future versions.

mvn23, 2022-11-16:
@PierreAronnax I have prepared a fix, but I am unable to test as my boiler supports the message ID. Would you be willing to help out with that?

PierreAronnax, 2022-11-17:
More than happy to help. I run Home Assistant on a Synology NAS in a Docker Container. Please tell me what to do.

mvn23, 2022-11-20:
As long as you have access to the files in the container we should be able to test.
All that needs to be done is to replace the existing `messageprocessor.py` and `messages.py` in the container with [this one](https://github.com/mvn23/pyotgw/blob/cff26816cce71123f43da0449f7fdce9dc5b3ab6/pyotgw/messageprocessor.py) and [this one](https://github.com/mvn23/pyotgw/blob/cff26816cce71123f43da0449f7fdce9dc5b3ab6/pyotgw/messages.py).
You may want to back up the existing files first to recover in case of problems. If the fix is OK the changed files should be overwritten automatically when HA is updated with the new pyotgw version.

PierreAronnax, 2022-11-20:
I have updated both files and can confirm it is working for me with Home Assistant 2022.11.2 and firmware 5.5 :)

Also, Schelte has released version [5.6](http://otgw.tclcode.com/changelog.html) that has also fixed this problem (tested that this morning, downgraded for this test)

PR #48 (2022-11-20): Fix handle TRSET on thermostat side
Fixes #46
closed 2022-11-20 by mvn23