mvn23/pyotgw #53: init_and_wait_for_activity fails on SE response

KJagersma, 2023-12-14:
In the home assistant opentherm gateway integration I ran into a setup problem, where the setup never succeeded. The source of the problem seems the be an SE response on the first PS request.

init_and_wait_for_activity has a retry of zero, however if the first response is SE, this results in an exception and a failed connect. Can this be changed > zero, or is there a workaround around this error possible?

see: https://github.com/home-assistant/core/issues/101837, I'm running an older version of the otgw (OpenTherm Gateway 4.2.4 / nodo shop hardware v2.0) connected with wifi.

I'm willing to test or create a pr if that makes it easier.

MedicoreV8, 2023-12-14:
Please make a PR request. Windows rebooted due to an upgrade and now the integration keeps failing to start.
I can no longer operate my thermostat within Home Assistant :(

mvn23, 2023-12-15:
I saw in the references Home Assistance issue that you've been digging around in the code a bit already. Could you test the following?

Change this line:
https://github.com/mvn23/pyotgw/blob/5b6741add9023cfbd83454f515adee154d7d9f37/pyotgw/commandprocessor.py#L41
to
```python
            self.protocol.transport.write(f"\r\n{cmd}={value}\r\n".encode("ascii"))
```

And a few lines down, this line:
https://github.com/mvn23/pyotgw/blob/5b6741add9023cfbd83454f515adee154d7d9f37/pyotgw/commandprocessor.py#L49
to
```python
                self.protocol.transport.write(f"\r\n{cmd}={value}\r\n".encode("ascii"))
```

The additional `\r\n` should make sure there's no leftover data on the serial line for the gateway to process. I've been planning to implement this for a while already, but never got round to actually testing it.

KJagersma, 2023-12-15:
Thanks for the feedback! This seems easy enough to test, but probably not before Sunday. I report back with the results as soon as possible.

KJagersma, 2023-12-17:
tldr: user error; not following instructions...

The root of the problem is me not following the setup instructions for my NodeMCU with ESPEasy:( After your suggested update didn't change anything for me, i noticed the first response was always SE, even without a command, and every 30 seconds an SE response would pop up. After some digging this is the result of leaving the serial log level to the default setting of 2 in the ESPEasy config. After changing this to zero, everything is working as intended again.
This is stated in the official NODO-shop setup guide, so totally my fault for not following instructions. I've used this setup in the past on Domoticz, so I wrongly assumed the configuration was ok.

Some other remarks while checking this issue:
- I think the pre command \r\n doesn't do what you want it to do: if there still is any garbage in the buffer of the gateway, this will be terminated and processed with the first \r\n, after which the intended command is processed, so you still receive an SE, or other error as a first response if there where some left over chars in the buffer.
- In the integration setup, the exception raised by an SE repsonse (SyntaxError) is not caught, so the gateway.cleanup section is never executed on this specific error, but I'm not sure this is a problem to worry about.
- Have you encountered this user error previously? If you think this would be a good addition, I can try to create a PR in the next holiday to report this specific error on setup, it seems easy enough to detect and it would have helped me tremendously if this was a detected error, instead of a generic exception. I haven't done anything with python or HA integrations, so it seems like a nice project to try.

Anyway, thanks a lot for your time! This issue can be closed as far as I'm concerned.

MedicoreV8, 2023-12-17:
Hold on a second, before closing it.
How do you change the Serial log level in the ESPEasy config?
Can this be done with a command in the terminal?

mvn23, 2023-12-19:
Should be on the web interface on the [url=https://www.letscontrolit.com/wiki/index.php/ESP_Easy_web_interface#Tools_Advanced_page]Tools Advanced page[/url] (reference may be outdated).

mvn23, 2023-12-19:
> tldr: user error; not following instructions...
> 
> The root of the problem is me not following the setup instructions for my NodeMCU with ESPEasy:( After your suggested update didn't change anything for me, i noticed the first response was always SE, even without a command, and every 30 seconds an SE response would pop up. After some digging this is the result of leaving the serial log level to the default setting of 2 in the ESPEasy config. After changing this to zero, everything is working as intended again. This is stated in the official NODO-shop setup guide, so totally my fault for not following instructions. I've used this setup in the past on openHAB, so I wrongly assumed the configuration was ok.
> 
> Some other remarks while checking this issue:
> 
>     * I think the pre command \r\n doesn't do what you want it to do: if there still is any garbage in the buffer of the gateway, this will be terminated and processed with the first \r\n, after which the intended command is processed, so you still receive an SE, or other error as a first response if there where some left over chars in the buffer.
> 
>     * In the integration setup, the exception raised by an SE repsonse (SyntaxError) is not caught, so the gateway.cleanup section is never executed on this specific error, but I'm not sure this is a problem to worry about.
> 
>     * Have you encountered this user error previously? If you think this would be a good addition, I can try to create a PR in the next holiday to report this specific error on setup, it seems easy enough to detect and it would have helped me tremendously if this was a detected error, instead of a generic exception. I haven't done anything with python or HA integrations, so it seems like a nice project to try.
> 
> 
> Anyway, thanks a lot for your time! This issue can be closed as far as I'm concerned.

Thanks for the feedback. There are indeed some issues with the solution I proposed. As you rightfully noted, it is not as simple as just adding the `\r\n` before sending the command.

It could be a good idea to add a readme-type page with instructions for specific firmwares for pyotgw. This could then for example be referenced in the Home Assistant documentation. Problem is that there are so many firmwares and versions out there so maintaining the document could become tedious in the long run, especially since I do not own a gateway with an ESP attached so I have no way to test those scenarios.

mvn23, 2023-12-19:
> Come to think of it, I don't have an esp attached to the gateway. I still have problems with loading the integration when rebooting hass.

Yours may be a different issue then. Please create a new issue and provide debug logs of a failing start.

closed 2023-12-19 by mvn23