mvn23/pyotgw #89: Missing opentherm_gw.set_target_temp (and others) in HA integration

blb4github, 2024-10-07:
Hi,

It looks like not all set options from the list in pyotgw are available in the Home Assistant integration, One I'm missing is set_target_temp. Would it be possible to get that one into the HA integration?

mvn23, 2024-10-07:
You can control the target temperature with the `climate` entity and its related `climate` service actions. What would be the use case for `set_target_temp` as a separate service action?
If you have a use case that is not yet covered by Home Assistant but is supported in `pyotgw` then the correct place would indeed be the HA core issue tracker.

closed 2024-10-07 by mvn23

blb4github, 2024-10-07:
Thanks, the answer for asking for this is I don't see the climate entity working. I'll have a further look at that first. Anyhow, this is then indeed more HA related then pyotgw.