mvn23/pyotgw #90: BS command to modify the room setpoint being sent to the boiler missing (version 6.5, PIC16F1847-only)

blb4github, 2024-10-07:
Hi,

since 6.5, PIC16F1847- only verion), a new command is added: 
BS=temperature
    Boiler setpoint. Some boilers ignore certain controls sent by the thermostat, based on the room temperature and room setpoint values. Sending a different room setpoint can help to convince these boilers to do what they are supposed to do. Set the temperature to 0 to return to passing along the value specified by the thermostat.
    Examples: BS=16.3, BS=0 

It would be great if this one can be added (and then to the HA integration as well) as with my Honeywell Evohome Opentherm thermostat the TT and TC commands don't work.

mvn23, 2024-10-07:
Thanks for the report. I have recently obtained a few PIC16F1847 controllers and plan to implement the additional commands in the near future.
I checked the documentation for the `BS` command and so far I don't find its explanation very clear. From what I understand it only affects the values sent to the boiler, so if the `TT` and `TC` commands don't work on your thermostat then the `BS` command won't change that. You can test this with otmonitor or in Home Assistant with the `opentherm_gw.send_transparent_command` [service action](https://www.home-assistant.io/integrations/opentherm_gw/#action-opentherm_gwsend_transparent_command).

If you're testing this anyway, I would appreciate a log of the protocol from otmonitor or a Home Assistant integration debug log with some information on the applied testing procedure if it's not too much trouble. Keep in mind that the opentherm protocol is slow and depends on the thermostat polling for values, so please give it some time (2-3 minutes should be enough, but longer could be necessary) to settle between commands when testing.

blb4github, 2024-10-07:
Thanks, I'll do the logging. The value provided with the BS command is indeed only send to the boiler, not to the Thermostat. My Opentherm Honeywell Evohome (multi-zone) thermostat is not setting the room temperature nor the required room temperature, which would be difficult indeed in a multi-zone installation. Currently I'm controlling my heatpump with chEnable, Control Setpoint and max ch water setpoint which should be fine but the behavior of my thermostat is not that smooth. I like to use the difference between my actual living room temperature and the living room setpoint in the logic and via the BS command I can sent this value to my heatpump.